BrianPugh/belay #8: WebREPL support?

roger-, 2022-08-14:
Is it possible to use this without a physically connected board (e.g. via the WebREPL)?

BrianPugh, 2022-08-14:
I haven't explicitly tested, but I'll work on this a little bit today. I'll update you on the results.

BrianPugh, 2022-08-15:
So i have a basic version working on the `webrepl` branch (see example 09). The performance is kind of lackluster, since I think a lot of the pyboard code from micropython was written under the assumption of a low-latency connection. I'll try and optimize it a bit, but mediocre performance might just be the best we get.

Any feedback would be appreciated!

BrianPugh, 2022-08-16:
Added support in the newly released `v0.4.0`, we can optimize performance over time, but this should be "good enough" for some projects.

closed 2022-08-16 by BrianPugh