BrianPugh/belay #14: No way to clean up the connection status / cleanly end a connection / cleanly restart a connection

pak-man, 2022-08-24:
I´d like to build on belay lib, but it lacks some mechanisms:
- how to call the status of a connection (connected, lost..)
- how to cleanly re-establish a connection

While i can use the exceptions (pyboard or serial exceptions) that a thrown to detect a problem, I cannot cleanly re-establish a connection with both resetting the board and restarting my script.

The idea is to automatically restart my belay task when after a devices is resetted or un-plugged > replugged (failproof connection)
(so far i tested only belay with serial connections)

BearXP, 2022-08-24:
> No way to ... cleanly end a connection

I noticed the same thing, I've been using the code:

```python
device = belay.Device("COM5")
# ... Processing ...
# Finished processing - close connection
device._board.close()
```

The `device_board.close()` seems happy enough to close it, so next time I run my code it makes a new connection.

BrianPugh, 2022-08-24:
In the next release, I'll make the `device._board.close()` method public at `device.close()`.

I'll try and figure out an API for more robust connection handlings in the coming few days.

BrianPugh, 2022-08-27:
`Device.close()` is now available in Belay v0.5.0.

If we want the main script to completely re-execute upon the device disconnecting, I think this can be easily done externally via some bash scripting.  However, if we have a daemon-like python program, we may want it to continue executing and attempt to re-establish connection with the device.

A challenge for re-establishing a connection with the device is that it's interpreter state will be lost, but this isn't super critical for all applications. For example, if the device is just acting as an interface for reading from sensors, there isn't much internal state. The following is a bit of a brain dump on the topic.

For the programming interface we need to record and replay (in-order) all previous device actions after reconnecting. Enumerated, I believe this is just:

1. Resend over all task/thread source code.
2. Re-invoke all commands (invoked via `__call__`).

This would all be encapsulated in a method, `device.reconnect()`. I think this would restore sufficient device interpreter state for most use-cases. I also think for most usecases, the number of tasks and commands would be static, but we would need to check that assumption so memory usage doesn't grow unbounded.

As for re-establishing the connection, we could either do this automatically or manually:

* Manual: the user catches an exception `LostConnection` and then invokes a `device.reconnect()`.
  Pros: safer, gives the user more control, nothing too unexpected.
  Cons: all belay command would need to be wrapped in a try/except, which might simply be too burdensome.
* Automatic: when an action is performed, internally catch `LostConnection` and attempt to invoke `device.reconnect()`, with some delay/backoff retry code. On my raspberry pi's, a dirty power-supply sometimes causes my pi-pico to intermittently disconnect. Up to 3 re-connect attempts with 5 seconds in-between attempts usually re-establishes a connection.
  Pros: User doesn't really have to worry about anything.
  Cons: Possibly unexpected blocking code.

So maybe a solution is a mixture of the two with Automatic being a configureable option. I'll have to think about it more, but would also love to hear your thoughts.

BrianPugh, 2022-08-28:
@pak-man I have a rough implementation in the [conn-retry](https://github.com/BrianPugh/belay/tree/conn-retry) branch. Would you like to give it a try? The summary is as follows:

1. Source code decorated by `task` and `thread`, along with some startup code are recorded into a command list. Basically the only calls that are **not** recorded are the ones called by actually executing task/thread.
2. At some point, the device resets/becomes disconnect and then reconnects to host. The REPL is basically in a new state.
3. Upon any call that sends a command to the device, an exception is raised because the device was previously disconnected.
4. If `attempts` provided to `Device.__init__` is set to a positive integer (e.g. `device = Device("/dev/ttyUSB0", attempts=10)`), then Belay attempts to automatically reconnect to the device up to that many times, with 1.0 seconds in-between attempts.
5. Upon reconnection,  replay recorded command list.
6. If connection couldn't be re-established raise `ConnectionLost` exception.

In my mind, the primary use of Belay is for interacting with sensors, actuators, and displays. Displays aside, I don't see there being much internal state, so I think this would benefit most use-cases. However, if anyone thinks otherwise, please let me know!

pak-man, 2022-09-06:
Sorry it totally missed your post, will give a try and give you feedback ASAP. I thank you a lot for your support and reactivity !

A bit off-topic, but lately I´ve been working with [pyodide](https://pyodide.org/en/stable/), an in-browser Python REPL with bindings to JS objects like the DOM, canvas, webGL (...). 

My first goal with pyodide is to be able to use Webserial Javascript API, then I´ll give a try a adapting or working Belay to get to work with pyodide + Webserial. Further goal is to provide an all-in-one, easily customisable solution for harvesting data from sensors hosted on some micropython boards, plotting, exporting without having to install anything, all running within the browser.
If you want to have a look, I´ve got a couple of notebooks [here](https://starboard.gg/ACBIAS/).

BrianPugh, 2022-09-06:
awesome! That sounds like a really cool, useful project! If you need any features from Belay, just let me know! I merged the mentioned `conn-retry` branch into `main`. I'm planning on cutting a release within the next day or two, so feedback is greatly appreciated!

pak-man, 2022-09-07:
Hello Brian,

I could test the connection retry feature and it works as expected. whenever I unplug or reset the board, my script resumes when the board is back. Having the belay lib "replay" the tasks on the board is a big advantage, at least for a scenario where board sensors are constantly read by the computer. 

Maybe just one remark that might fit in the docs: 
- Many time when one unplugs-plus a board, it get a different device name (like /dev/ttyUSB0 is then /dev/ttyUSB1) in this case, the belay lib cannot reconnect. 
- A possible mitigation is to use udev, and symlink an alias to the device name. 
- As example i created `/etc/udev/rules.d/99-mypicoboard.rules` that contains:
`SUBSYSTEMS=="usb", KERNEL=="ttyACM[0-9]*", ATTRS{idVendor}=="2e8a", ATTRS{idProduct}=="0005", ATTRS{serial}=="e6616407e3000021", SYMLINK+="mypicoboard", MODE:="0666"` : The board with this serial number will always be accessible under /dev/mypicoboard symlink.
- To find out which serial number has the board the command `udevadm info -a -n /dev/ttyACMx | grep serial` is used, where /dev/ttyACMx (could be also /dev/ttyUSBx) is my board.
- The usual `sudo udevadm control --reload-rules && sudo udevadm trigger` refreshes the udev rules without having to reboot
- This is valid for the RP2040 but might apply to some other USB-TTL bridges, however not all provide a unique serial number.

closed 2022-09-07 by pak-man

BrianPugh, 2022-09-08:
Added udev rules to the docs in #24. Will release v0.7.0 very soon.