BrianPugh/belay #15: Yield result

BearXP, 2022-08-24:
First I just wanted to thank you for this library, it's been super helpful!

I've just run in to a problem, I would like to yield a result instead of returning a result.

This is works fine on my pi pico:
```python
from time import sleep

def count():
    i = 0
    while True:
        i += 1
        yield i

for index in count():
    sleep(1)
    print(index)
```

However when I try and use belay with:
```python
import belay
from time import sleep

device = belay.Device("COM5")

@device.task
def count():
    i = 0
    while True:
        i += 1
        yield i

for index in count():
    print(index)
```

I receive the error:
```Argument of type "() -> Generator[Literal[1], None, None]" cannot be assigned to parameter "f" of type "((...) -> PythonLiteral) | None" in function "__call__"
  Type "() -> Generator[Literal[1], None, None]" cannot be assigned to type "((...) -> PythonLiteral) | None"
    Type "() -> Generator[Literal[1], None, None]" cannot be assigned to type "(...) -> PythonLiteral"
      Function return type "Generator[Literal[1], None, None]" is incompatible with type "PythonLiteral"
        Type "Generator[Literal[1], None, None]" cannot be assigned to type "PythonLiteral"
          Type cannot be assigned to type "None"
          "Generator[Literal[1], None, None]" is incompatible with "bool"
          "Generator[Literal[1], None, None]" is incompatible with "bytes"
          "Generator[Literal[1], None, None]" is incompatible with "int"
```

Thanks!

pak-man, 2022-08-24:
I agree this would be nice to have,and perfectly fit many belay use cases !

BrianPugh, 2022-08-24:
Thanks for the interest! I should have some time in a few days to work on this, hopefully I'll figure out a way to support generators.

BrianPugh, 2022-08-27:
@pak-man @BearXP checkout the newly pushed `generators` branch. I still have to polish it up, but see if it works for your use cases (I added a 10th example using your demo code in this issue). I'm going to try and merge (and do a release) sometime over the next day or two.

BrianPugh, 2022-08-27:
Belay v0.5.0 has been released, which includes this generator support. Closing this issue for now. We can re-open/start a new issue if you find this hasn't been fully resolved. Thanks!

closed 2022-08-27 by BrianPugh