BrianPugh/belay #40: Advised way of importing libraries

roaldarbol, 2022-11-17:
I was wondering how you can import libraries other than the ones imported by default by Belay. I've made a `@device.task def setup():` function which works well, but is that the intended way? As a side note I'm awaiting Micropython v1.20 which includes `mip` (so you should be able to install using `mip install <package>` if I've understood correctly). Maybe it would be a nice addition  to have a "installing and importing libraries" section in the documentation?

BrianPugh, 2022-11-17:
so there are multiple ways, here are some thoughts in no particular order:

1. Belay [now contains a package manager](https://belay.readthedocs.io/en/latest/Package%20Manager.html), so you can manage libraries externally from your script. This is (obviously) my preferred way for the reasons outlined in the documentation. This syncs files into the  on-device `lib` folder.

2. It might be nice to programmatically run this in your script; should I expose a `device.install()` method that basically performs the `belay install` action? I should probably also add a `device.update()` method as well, but thats less critical since this should probably be manually ran anyway.

3. Currently, the best way of moving local files over is using the `device.sync` method, but that might be made (relatively) obsolete with the proposed `device.install` method mentioned in (2) for the common case of moving over your project's package.

4. As for actually importing libraries into the global scope, currently you have to do `device("import my_library")`. Running imports I could add a tiny bit of syntactical sugar to make it look like `device.import("my_library")`, but I don't think it's necessary to make the interface a little more complicated without making the importing process actually any easier. Importing libraries within a task *will not* make them available in the global scope, as its equivalent of the following (run in repl):

```
>>> def setup():
    import os
    
>>> 
>>> setup()
>>> dir(os)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'os' isn't defined
```

roaldarbol, 2022-11-17:
The package manager is already dawning on me. I really really like it! However, I'm a bit confused by:

> Belay assumes your project contains a python-package with the same name as tool.belay.name located in the root of your project.

Does that mean you cannot have just a `pyproject.toml` file, the python script and nothing else in a repo? (except for the autogenerated `.belay-lib)

BrianPugh, 2022-11-17:
as for your proposed `device.import`, thats actualy just the same as the normal `device.__call__` method; i.e.:

```
device("import logging; from neopixel import Neopixel")
```

All `device("python statement here")` are executed on-device in the global context, so it's useful for imports and declaring global variables/objects. A possible workflow is instantiating hardware this way, and then using it inside tasks. An alternative is your `setup`, but no need to make it builtin, you could do something like:

```
@device.task
def setup():
    from mysensor import MySensor
    global my_sensor
    my_sensor = MySensor(Pin(15))
    
setup()
```

As a design philosophy, I'm trying to keep the number of `device` methods as minimal as possible (whole zen of python thing). I'd only feel inclined to add additional methods if it makes the user's life significantly easier (e.g. `device.sync`). Since both setup, library import, and global declarations can be done without much effort using the existing toolset, I don't feel inclined to add more methods.

Thinking about it more, there is one missing functionality here. Belay has the ability to replay commands, which is useful if a device disconnects/resets for whatever reason. Currently all task executions are **not** recorded. I should add an option to the task/thread decorator to allow recording, something like:

```
@device.task(record=True)
def setup():
    from mysensor import MySensor
    global my_sensor
    my_sensor = MySensor(Pin(15))
    
setup()
```

this way, your `setup()` will be re-executed upon re-connect.

roaldarbol, 2022-11-18:
For some reason I have a great dislike for using `global`, simply because it adds extra lines. A way around it could be something suggested on [StackOverflow](https://stackoverflow.com/questions/11543297/make-all-variables-in-a-python-function-global), adding `globals().update(locals())` to the end of the function to make everything available globally. Here I'd love a `@device.global` that adds that line to the call itself, rather than having to write it yourself at the end of a `@device.task`. I get what you mean about keeping methods minimal, but I'd also argue that pythonian philosopy also says to make things *explicit* rather than *implicit*. And using just `device()` is a very implicit way of going about things I'd argue.

BrianPugh, 2022-11-18:
how about something like:
```
@device.task(record=True, global=True)
def setup():
    from mysensor import MySensor
    my_sensor = MySensor(Pin(15))
    
setup()
```

BrianPugh, 2022-11-18:
so i was looking at how this would be implemented, and I think there are too many sharp corners for it to work. Example:

```
device("foo = 5")

@device.task(global=True)
def setup():
    foo += 1

setup()
```

This wouldn't work as expected. As far as implicitness, I think nothing is wrong using `device()`, the functionality is very explicit: execute code on device.

roaldarbol, 2022-11-21:
Maybe this could be amended by solving #51? If the user can create a class, then a `setup` function would automatically update variables globally within the object. I.e. when this kind of work flow, we could recommend creating a class.

BrianPugh, 2022-11-29:
@roaldarbol checkout #54 and provide feedback in the PR. I'm going to close this issue for now.

closed 2022-11-29 by BrianPugh