BrianPugh/belay #43: Package manager dependency indexes

roaldarbol, 2022-11-17:
I've not used the package manager yet, but just based on the documentation it seems that you have to specify URLs to repos. Poetry indexes PyPi, so maybe it could be an idea to do something similar to automatically index both PyPi and micropython-lib? E.g. having
```toml
[tools.belay.dependencies.pip]
toml = "^0.10.2"

[tools.belay.dependencies.mip]
logging = "^0.5.2"
```
or a more Poetry compatible version:
```toml
[tools.belay.group.pip.dependencies]
toml = "^0.10.2"

[tools.belay.group.mip.dependencies]
logging = "^0.5.2"
```

And since Micropython packages can also be in places like [pycopy](https://github.com/pfalcon/pycopy-lib) maybe thinking about how to implement specify indexes more generally (like secondary indexes in Poetry) would be worthwhile. 😊

BrianPugh, 2022-11-17:
The current issue is that there isn't a strong centralized micropython package repository. Frequently there are just isolated independent github repos/gists. Worse still, a lot of micropython projects don't have versioning, so thats also a bit of a mess. There is some stuff on PyPi (with the `micropython-` prefix), but I'm not sure how strong/global that effort is. That said, the current URL approach just seemed like the easiest, globally useful solution (that can also be easily extended upon).

Some of these issues are being recently addressed by[ mip's `package.json`](https://docs.micropython.org/en/latest/reference/packages.html#writing-publishing-packages), but I'm not entirely convinced by it yet.

Secondary indexes could be implemented, but I feel like the syntax would be *almost* as verbose as just pasting the link directly to the file.

I'm still very open to ways to improve this, just looking for a fully (or mostly) fleshed out idea before implementing.

roaldarbol, 2022-11-18:
I think it will be a super way of getting people to use Belay. Exactly *because* package management is not a big thing **yet**! I think package managers have not been widely used because there hasn't been a good way to work with them; either you could use Thonny or `upip`, but only if you had networking capabilities. Having an easy way to make use of the package managers is gonna bring people on board, and in turn make them more appealing. I really think there's something there, but I do agree, the idea should be properly fleshed out before implementing something. It will also make people less dependent on supplier-specific Micropython firmware (e.g. Adafruit/Pimoroni), if they can be convinced to upload their libraries to a package manager (I did see some of Pimoroni's packages on PyPi already I think).

BrianPugh, 2023-03-16:
I'm gonna close this for now. We could potentially make some of the configuration nicer by doing something like `[[tool.poetry.source]]`, but I'm going to hold off on implementing that until someone actually, justifiably, needs it.

closed 2023-03-16 by BrianPugh