BrianPugh/belay #58: AttributeError: type object 'Executer' has no attribute 'setup'

roaldarbol, 2022-11-30:
Just a bug report. I'm trying to use the new `@Device.setup` decorator (I am on v0.14.1), but I get the following error:
```python
from belay import Device
import time

class PicoW(Device):
    @Device.setup
    def setup():
        led = Pin('LED', Pin.OUT)

    @Device.task
    def led_toggle():
        led.toggle()

if __name__ == "__main__":
    picoW = PicoW('/dev/cu.usbmodem14240')
    picoW.setup()
    picoW.led_toggle()
    time.sleep(2)
    picoW.led_toggle()
    picoW.close()

# Traceback (most recent call last):
#   File "/Users/roaldarbol/MEGA/Documents/sussex/research/experiments/experiments/PicoW.py", line 14, in <module>
#     picoW = PicoW('/dev/cu.usbmodem14240')
#   File "/usr/local/lib/python3.10/site-packages/belay/device.py", line 251, in __init__
#     executer = getattr(Executer, metadata.executer.__registry__.name)
# AttributeError: type object 'Executer' has no attribute 'setup'
```

BrianPugh, 2022-11-30:
Fixed in #59 and `v0.14.2`. Don't know how that slipped through, sorry about that!

roaldarbol, 2022-11-30:
Haha, no apologies needed! I’ll check later🤘🏼