BrianPugh/belay #62: `Yield` returns `None`

roaldarbol, 2022-12-02:
I think there may have been a breaking change somewhere. I'm trying a simple generator function, but I only get `None` yields. Tried on both my Mac and a student's Windows with Belay `0.15`.
```python

from belay import Device
import time

class PicoW(Device):
    @Device.task
    def my_generator():
        i = 0
        while i < 10:
            yield i
            i += 1

if __name__ == "__main__":
    picoW = PicoW(list_devices()[-1])
    for val in picoW.my_generator():
        print(val)
        time.sleep(1)
    picoW.close()
```
If I print from within the generator I get the correct values, so I believe they're lost during transit somewhere.
I also tried in the old-fashioned way without a class, getting the same results.

roaldarbol, 2022-12-02:
Maybe we'll need to add a test asserting generator output?

BrianPugh, 2022-12-02:
I accidentally broke this in f7244cbe6d7816ee54f8ba96bf92c20f4acc816f. What happened is that the belay-internal function used for getting the next element of a generator was being interpreted as a generic expression, which lead to some downstream issues. This is fixed in 33fe6dc18159288d0ee2303cb0f91f8a2885b3c0 and release `v0.15.1`.

The fix is a little bit of a hack, but it's a very minimal change. The proper code should be a little bit simpler since that commit fixed expressions. Since broken code is becoming an issue, I definitely want to setup more integration tests before adding new features/more refactoring. I've been avoiding it a bit at this point because its not exactly easy to setup for this. I'm unavailable for a few days, but will work on it when I get back.

closed 2022-12-02 by BrianPugh

roaldarbol, 2022-12-02:
Thanks! Checked and works for me👌

BrianPugh, 2022-12-06:
Setup and added integration tests in #63 , so that this shouldn't happen in the future (and if breaks do happen, we can now easily add integration tests to prevent regressions).