BrianPugh/belay #66: Using Pimoroni libraries

roaldarbol, 2022-12-12:
...this is a tentative issue title, until we figure out what the underlying problem should be. Let's see. ;-)

I'd like to be able to use Pimoroni's sensors, and they have lots of libraries for them (found [here](https://github.com/pimoroni/pimoroni-pico)). However, I can't find the appropriate `.py` files, only `.cpp` or `.hpp`. Any ideas about how to use those? If there's an easy way of using those, then the issue is just this.

They have made their own micropython firmware which includes all their libraries. I tried to just upload their firmware on my Pico, but when loading the libraries I can't (`NameError: name 'pimoroni_i2c' isn't defined`). So I guess we don't have access to the on-board libraries - or am I just doing something wrong? 

So, perhaps the issue could also be about how to work with alternative firmware. I thought there might be a case for having a section in `pyproject.toml` where one could specify a URL to a custom firmware, and if it isn't already loaded, then load that firmware onto the board? I don't know if it'll be confusing, but it would help with "working out of the box".

However, I think it can become problematic with multiple boards, and we would have to put some thought into it. 

Once `mip` flourishes, it should possibly become redundant, but for now I think there's a need.

BrianPugh, 2022-12-12:
it seems like most of pimoroni's stuff is in cpp, and they have python bindings. After you flash their firmware, can you run

```
help('modules')
```

in the device's terminal?

I'd thing install-firmware-for-you is a good idea; it just gets complicated because we then need to keep a bunch of code on how to flash different boards. We could maybe just support the common ones (espressif chips, pi pico, etc).

roaldarbol, 2022-12-13:
Thanks, good idea! So the modules are on there, I get:
```zsh
__main__          breakout_pmw3901  neopixel          uasyncio/stream
_boot             breakout_potentiometer              onewire           ubinascii
_boot_fat         breakout_rgbmatrix5x5               pcf85063a         ucollections
_onewire          breakout_rtc      picoexplorer      ucryptolib
_rp2              breakout_scd41    picographics      uctypes
_thread           breakout_sgp30    picokeypad        uerrno
_uasyncio         breakout_trackball                  picoscroll        uhashlib
adcfft            breakout_vl53l5cx picounicorn       uheapq
breakout_as7262   builtins          picowireless      uio
breakout_bh1745   cmath             pimoroni          ujson
breakout_bme280   dht               pimoroni_bus      umachine
breakout_bme68x   ds18x20           pimoroni_i2c      uos
breakout_bmp280   encoder           plasma            urandom
breakout_dotmatrix                  framebuf          qrcode            ure
breakout_encoder  galactic          rp2               uselect
breakout_icp10125 gc                servo             ustruct
breakout_ioexpander                 gfx_pack          uarray            usys
breakout_ltr559   hub75             uasyncio/__init__ utime
breakout_matrix11x7                 jpegdec           uasyncio/core     uzlib
breakout_mics6814 math              uasyncio/event
breakout_msa301   micropython       uasyncio/funcs
breakout_paa5100  motor             uasyncio/lock
Plus any modules on the filesystem
```
And just double checked, the error is still there. And I am able to import from Thonny.

I like the idea of just managing a few common types of firmware. And users could then suggest new ones - once we have them on our list, we could make an action that checks for new releases every now and then, or simply always fetch the latest release (or specified version?).

BrianPugh, 2022-12-13:
I'll put the firmware thing on the list of features to add.

As for solving your actual problem, it seems like your board is setup fine. Can you show me your Belay code?

roaldarbol, 2022-12-14:
Cool! And lol. It was just a dumb coding error on my side - works perfectly now. 🤦‍♂️

closed 2022-12-14 by roaldarbol

yezzer, 2023-03-06:
> Cool! And lol. It was just a dumb coding error on my side - works perfectly now. 🤦‍♂️

I'm having problems with importing Pimoroni modules too, how did you do this in the end? :)