BrianPugh/belay #67: Timed generator helper function

roaldarbol, 2022-12-13:
Hey Brian. I'm finding myself using the generators as my go-to currently, and started working on a single function to ease with logging data. So something like (roughly copy-pasted out of context):
```python
try:
        for val in device.my_generator(iti):
            val_pd = pd.DataFrame([val])
            data = pd.concat([data, val_pd], ignore_index=True)
            data.loc[data.index[-1],'time'] = time.time() - t_init
            print(val_pd)
            t0 = time.time()
            t1 = time.time()
            iti_adjusted = iti - ((t0 - t_init) % iti)
            while t1 - t0 < iti_adjusted:
                t1 = time.time()
    except:
        pass
```
Could instead be something like: 
```python
values = timed_generator(task, repeats, interval_type, duration, **args, **kwargs)
```

So I was wondering if that would be something it could make sense to include Belay or if I should make it somewhere separate? I am currently using `pandas` which is a major dependency to add onto the project. On the other hand it might be nice to have a few functions to aid various routine tasks, maybe in a separate sub-library. What do you think? I can create a pull request later this week if you'd like to see a draft.

roaldarbol, 2022-12-13:
I'm thinking it may be better to have it in a new library (e.g. `autobelay`), which can then take on heavier dependencies and focus more on the logging and post-processing...

BrianPugh, 2022-12-13:
yeah, I think this is starting to get outside the scope of Belay; I can see the general use, but I worry it might make the API seem complex or the dependencies too heavy.

Semi-related: inside your loop, you shouldn't call `pd.concat`. Doing so is `O(n^2)`; you should ideally just append to a list in your loop, and call a single `concat` or `stack` outside of the loop once data acquisition is complete.

roaldarbol, 2022-12-14:
Yeah, I thought so. I'll start a new repo instead - will be happy to discuss best practices there too! 👌

The reason I did it that way is that I `yield` a dict so I can keep multiple variables alongside variable names for clarity - that just meant that appending became unwieldy. And I couldn't find a way of appending within a dict (it's always appending a new key/value pair); hence the `pd.concat` solution. If you have ideas about how to accomplish that, I'd be happy to hear them!

The reason for the loop is that, when timing is really important, comparing times is preferable as running the code on the board has some variability in duration which you cannot account for using `time.sleep`. Say you want to run something every 5 seconds and the code takes between 0.01-0.05 seconds - that builds up quite quickly. `time.sleep` is definitely good in some use cases though. I traditional learning experiments with rodents they use something called the Inter-trial Interval (ITI), i.e. "end-to-start" interval, which is exactly what `time.sleep` does. But a "start-to-start" interval is often necessary, and that's the use case for the loop. 😊

closed 2022-12-14 by roaldarbol