BrianPugh/belay #68: Injecting methods

roaldarbol, 2022-12-14:
Again, just running an idea by you to hear your thoughts. Would it be possible to inject calls to the `close` method and the class instantiation somehow? 

Is there a way one can check if any functions with a particular decorator is present? Then, upon instantiating the `class`, we could check to see whether there are any `@Device.setup` methods and automatically run them, and similarly, with a `@Device.close`, inject them into the default `close` method?

For close, the way I work right now, which is still super easy is just to make a new task of some name, say `housekeeping` which turns off LEDs, stops motors, etc. I just thought that it would be more intuitive if you could inject them into `close`, so you know that all the housekeeping will always be performed when closing the device.

E.g. in this minimal example, the `led` would be automatically set up, and turned off:

```python
class Pico(Device):
    @Device.setup
    def setup():
        led = Pin(25, Pin.OUT)
      
    @Device.task
    def led_on():
        led.on()

    @Device.close
    def close():
        led.off()

if __name__ == "__main__":
    pico = Pico(list_devices()[-1])
    pico.led_on()
    time.sleep(3)
    pico.close()
```

BrianPugh, 2022-12-14:
I think thats a good idea that should be pretty easy to implement (can be implemented soon).

Naming-wise, we have options:

1. `close` - issue is is that it might confuse people since `device.close()` also exists. Conversely, it might also make it easy to remember since it's the same name. This hasn't stopped us before with `task`/`thread`, but the difference here is that it's semantic meaning is different. Another slight inconsistency is that these cleanup tasks are automatically executed on `close`, but the `setup` method isn't automatically executed on object creation. This was done intentionally so that arguments could be passed to it, and lengthy setup calls could be done at a different location in code.
2. `atexit` - similar to the [python builtin](https://docs.python.org/3/library/atexit.html).
3. `exit` - same as (2), but reads off nicer due to the `@` in the decorator.
4. `teardown` - similar to the [python builtin unittest](https://docs.python.org/3/library/unittest.html#unittest.TestCase.tearDown). This pairs nicely with the existing `setup`.
5. `cleanup` - Just another option, don't like this one too much.

roaldarbol, 2022-12-14:
Cool!

For setup, could there be a way of specifying `autoload=True` somehow in a setup call?

Naming wise, I agree it might be confusing to also use close for such a method. I like `exit`!

BrianPugh, 2022-12-14:
cool, lets go with `teardown` then.

For `setup`, we can definitely add a parameter like `autoload`. It would only be valid if there are no arguments in the decorated function signature.:

```
class Foo(Device):
    @Device.setup(autoload=True)
    def setup():
        pass
```

Same exercise here, what would we like to call it?

1. `auto=True` - short, but maybe too short.
2. `autouse=True` - same as `pytest` fixtures.
3. `autoload=True` - another fine name, but `load` might confuse someone (does it load the code onto the device?).
4. `init=True` - implies it gets called on object initialization.
5. `autoinit=True` - Same as (4)

roaldarbol, 2022-12-14:
Awesome! Didn’t know @ decorators could take actual input like that - pleasantly surprised.😃

I like the `init` solutions as they refer the the timat which it will be called. Think I prefer `autoinit` slightly, but don’t quite know why😅

BrianPugh, 2022-12-14:
alright, lets go with `autoinit=False`. I'll implement these features sometime over the next few days.

roaldarbol, 2022-12-14:
Sounds good! Looking forward to testing it out!

BrianPugh, 2022-12-16:
See #69 . I still need to update docs/examples.

BrianPugh, 2022-12-18:
`teardown` has been introduced in #70. I'll be doing a release probably tomorrow, I want to wrap up a few other small features.

closed 2022-12-18 by BrianPugh

BrianPugh, 2022-12-19:
v0.16.0 has been released which includes these new features.