BrianPugh/belay #76: Send event/request from board/device

emcek, 2023-01-11:
Hi,
I got find `belay` in search of way to communicate RPi Pico <=> PC via serial (both ways).
I saw examples (i.e. 07_lcd), and it looks like communication is initiate by PC only.
Beside printing something at LCD or return value from some function executed at Pico, I need as well way to get notified about pushed button at Pico (or event form LCD touch screen).
Shortly when message/request was initiate by Pico itself.

Can `belay` support this, or maybe you know already existed solution.

BrianPugh, 2023-01-11:
Hi @emcek! As you noticed, all transactions in Belay are initiated by the  PC. There are a few less-than-perfect ways to do this with Belay. I think the best solution would be to have an interrupt on-button-press to set a flag on-device, and periodically poll for this flag from the PC. You could also directly read the input while polling from PC, but you might miss rapid button presses that way.

I think that this is a common enough use-case that if we can figure out a generic way of making this easier, I'd be happy to add features to Belay to support it.

emcek, 2023-01-11:
Yes, I got same idea... with setting a flag, my project is at early stage so maybe delay wouldn't be an issue.

BrianPugh, 2023-01-11:
please keep me updated with the route you end up taking and your success with it!

freemansoft, 2023-01-30:
Micropython doesn't support multiple CDC on USB https://forums.raspberrypi.com/viewtopic.php?t=305834 You could use a 2nd UART on the microcontroller to create a connection between the microcontroller and your PC.  Belay would continue to use the primary to interact.  Your microcontroller would write data to that 2nd UART so that the host program could read it.  https://docs.micropython.org/en/latest/library/machine.UART.html 

1. You could connect the PC and controller using a Serial/USB cable, a second wired collection.  
2. I've done something similar on a UART (not in belay) with a Bluetooth CDC virtual serial.

Another option is to use CIrcuitPython and USB CDC to add a 2nd USB serial port. You could open a 2nd virtual serial port using usb_cdc in circuit python.  https://docs.circuitpython.org/en/latest/shared-bindings/usb_cdc/index.html

I'm just spitballing here and haven't tried it yet.

BrianPugh, 2023-02-22:
closing now for inactivity, I'll re-open if there's more discussion.

closed 2023-02-22 by BrianPugh