BrianPugh/belay #77: could not get source code error

freemansoft, 2023-01-20:
Trying to run the LED example on an RP Pico

 

``` error log
PS C:\Users\joe\Documents\GitHub\belay>
>>> parser = argparse.ArgumentParser()
>>> parser.add_argument("--port", "-p", default="COM21")
_StoreAction(option_strings=['--port', '-p'], dest='port', nargs=None, const=None, default='COM21', type=None, choices=None, required=False, help=None, metavar=None)>>> args = parser.parse_args()>>>>>> # Setup the connection with the micropython board.
>>> # This also executes a few common imports on-device.
>>> device = belay.Device(args.port)
>>>>>>... def setup():  # The function name doesn't matter, but is "setup" by convention....     from machine import Pin
...     print('hello world')...Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "C:\Users\joe\Documents\GitHub\belay\belay\executers.py", line 57, in __call__
    src_code, src_lineno, src_file = getsource(f, strip_signature=True)
  File "C:\Users\joe\Documents\GitHub\belay\belay\inspect.py", line 91, in getsource
    lines, src_lineno = inspect.getsourcelines(f)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\lib\inspect.py", line 1006, in getsourcelines      
    lines, lnum = findsource(object)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\lib\inspect.py", line 835, in findsource
    raise OSError('could not get source code')
OSError: could not get source code
>>>
``

### Code
This code blows up on `print(Hellow World)`
```
import argparse
import time

import belay

parser = argparse.ArgumentParser()
parser.add_argument("--port", "-p", default="COM21")
args = parser.parse_args()

# Setup the connection with the micropython board.
# This also executes a few common imports on-device.
device = belay.Device(args.port)

@device.setup
def setup():  # The function name doesn't matter, but is "setup" by convention.
    from machine import Pin
    print('hello world')
```

BrianPugh, 2023-01-20:
I'm assuming you're attemting to define the task in a REPL. Belay works by using introspection to get the source code of task-decorated functions/methods. It does this by using the `getsourcelines` function of the builtin `inspect` module. The issue is described in [more detail here](https://stackoverflow.com/questions/57035364/inspect-getsourcelinesobject-shows-oserror-if-run-from-python-shell-but-gives).

If you run the code from a file, it should work as expected.

freemansoft, 2023-01-20:
Yupl  That was it. I"ve gotten used to testing in the REPL and didn't think about the introspection or stub generation or that type of thing.

closed 2023-01-20 by freemansoft

BrianPugh, 2023-01-20:
side note: glad to see its working on Windows, I haven't really done any Windows testing 😄