BrianPugh/belay #78: sync uses wrong leading path characters when copying files to /pyboard while running on windows machines.

freemansoft, 2023-01-21:
Was testing examples/06_external_modules_and_files with REPL across a Bluetooth Serial connection.  The sync operator appears to be doing something wrong when copying from the windows filesystem to the device file system.

Test machine: Windows PC
Target device: Pico RP2040

Symptoms:
1. Test program copies files that somehow end up prefixed with the backslash character  probably the windows root separator.
1. Program fails to find led.py 

Board directory after test fails.
```
C:\ESP8266-MicroPython> ls /pyboard/
**\**somemodule/       **\**led.py            **\**somemodule**\**led.py  **\**hello_world.txt
```

Board directory after manually restoring boot.py into same device
```
C:\ESP8266-MicroPython> ls /pyboard/
\somemodule/       \led.py            \somemodule\led.py **boot-orig.py**       \hello_world.txt
```

Expected results
```
C:\ESP8266-MicroPython> ls /pyboard/
somemodule/       led.py            somemodule/led.py   hello_world.txt
```

BrianPugh, 2023-01-21:
That's certainly an issue! I'll try and add Windows to the CI tests so I can catch and fix these. I should be able to get to it later this week.

BrianPugh, 2023-01-27:
just updating to let you know that I'm working on this; the fix should be fairly simple, the harder part is just to get all the windows CI working so that I can confidently support windows/fix other issues that may come up. Should have a release done by end of weekend.

freemansoft, 2023-01-28:
Thanks for the update
https://joe.blog.freemansoft.com/2023/01/an-open-source-developer-gave-best-in.html

BrianPugh, 2023-01-29:
Should be fixed and addressed in belay `v0.16.1`. CI now additionally runs on a Windows environment and all unit/integration tests pass. My windows dev environment isn't the best, so if you notice other Belay issues, please open up another Issue!

freemansoft, 2023-01-29:
Should I be able to verify this using the main branch?  `git pull`
I see the same behavior if I run the following on my windows machine
```
cd examples/06_external_modules_and_files
python3 main.py -p <COMPORT>
```

BrianPugh, 2023-01-29:
it's working for me; are you sure you're using the absolute latest? What does `python3 -c "import belay; print(belay.__version__)"` output? If you previously `pip install belay`, it might be using an older version rather than the local git copy that you think it might be.

freemansoft, 2023-01-29:
```
PS C:\Users\joe\Documents\GitHub\belay\examples\06_external_modules_and_file_sync> python3 -c "import belay; print(belay.__version__)"
0.16.0
```

pip uninstalled belay then reinstalled

```
Python 3.9.13 (tags/v3.9.13:6de2ca5, May 17 2022, 16:36:42) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import belay
>>> print(belay.__version__)
0.16.1
>>>
```

BrianPugh, 2023-01-29:
Is it working now that you're on 0.16.1?

freemansoft, 2023-01-29:
Yes.  Sorry about the false alarm.

Still seeing that I have to do a reset to get the repl back after running 

```
PS C:\Users\...\Documents\GitHub\belay\examples\06_external_modules_and_file_sync> python3 main.py -p COM4     
Using synced "led.py" via explicit commands.
Using synced "led.py" via task decorator.
Using synced "somemodule/led.py" via task decorator.
Reading a synced file via task decorator. Contents:
Hello World
The Quick Brown Fox Jumped Over The Lazy Dog.

PS C:\Users\...\Documents\GitHub\belay\examples\06_external_modules_and_file_sync> rshell -p COM4
Using buffer-size of 32
Connecting to COM4 (buffer-size 32)...
Trying to connect to REPL .
```
hangs there

freemansoft, 2023-01-29:
Still seeing that I have to do a reset to get the REPL back after running example 6

```
PS C:\Users\...\Documents\GitHub\belay\examples\06_external_modules_and_file_sync> python3 main.py -p COM4     
Using synced "led.py" via explicit commands.
Using synced "led.py" via task decorator.
Using synced "somemodule/led.py" via task decorator.
Reading a synced file via task decorator. Contents:
Hello World
The Quick Brown Fox Jumped Over The Lazy Dog.

PS C:\Users\...\Documents\GitHub\belay\examples\06_external_modules_and_file_sync> rshell -p COM4
Using buffer-size of 32
Connecting to COM4 (buffer-size 32)...
Trying to connect to REPL .
```
hangs there

freemansoft, 2023-01-29:
This issue is closed. I can open another one about the REPL loss in Example06

closed 2023-01-29 by freemansoft

BrianPugh, 2023-01-30:
I've got a fix incoming for that as well.

BrianPugh, 2023-01-30:
Fixed now in Belay v0.16.2