BrianPugh/belay #93: Unable to synch with CircuitPython 8.0 on Trinkey

freemansoft, 2023-02-09:
Unable to run examples/02_blink_neopixel against an adafruit board running CircuitPython 8.0 release. Haven't tried on 7.x

https://circuitpython.org/board/neopixel_trinkey_m0/

```
PS C:\Users\joe\Documents\GitHub\belay\examples\02_blink_neopixel> python3 .\circuitpython.py -p COM12
Traceback (most recent call last):
  File "C:\Users\joe\Documents\GitHub\belay\examples\02_blink_neopixel\circuitpython.py", line 12, in <module>
    device = belay.Device(args.port)
    self._connect_to_board(**self._board_kwargs)
  File "C:\Users\joe\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\belay\device.py", line 337, in _connect_to_board    self._board.enter_raw_repl(soft_reset=soft_reset)
  File "C:\Users\joe\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\belay\pyboard.py", line 421, in enter_raw_repl    self.read_until(1, b"soft reboot\r\n")
  File "C:\Users\joe\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\belay\pyboard.py", line 398, in read_until  
    raise PyboardError(
belay.pyboard.PyboardError: Timed out reading until b'soft reboot\r\n'
    Received: b'OK\r\nraw REPL; CTRL-B to exit\r\n>'

```

BrianPugh, 2023-02-09:
does circuitpython 8.0 work on the pi pico? Or is this a Trinkey-specific issue?

freemansoft, 2023-02-09:
It looks like it does.

https://circuitpython.org/board/raspberry_pi_pico_w/

I’ll load it on one of my pico’s to check.

BrianPugh, 2023-02-09:
I just tested circuitpython 8.0.0 on one of my rp2040 boards; its definitely a generic circuitpython 8.0.0 issue. I previously didn't experiment with circuitpython 8.0.0 since it was only in beta, but it looks like it got an official release 3 days. Since there's an official release now, I'll try and remedy this issue soon.

BrianPugh, 2023-02-09:
Digging a lil deeper. This is because circuitpython no longer prints `soft reboot` whenever the reboot ctrl-d signal is sent:

Circuitpython 7.3.3
```
Adafruit CircuitPython 7.3.3 on 2022-08-29; Raspberry Pi Pico with rp2040
>>>
>>>
>>>
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

Circuitpython 8.0.0
```
Adafruit CircuitPython 8.0.0 on 2023-02-06; Raspberry Pi Pico with rp2040
>>>
>>>
>>>
>>>
␛]0;🐍Done | 8.0.0␛\
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

Will work on a solution, should be fairly easy.

freemansoft, 2023-02-09:
### CircuitPython 8
```
Circuit Python Serial Monitor
[Open] Connection to COM12

press Ctrl-C to enter the REPL

8.0.0Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### CircuitPython 7
````
Circuit Python Serial Monitor
[Open] Connection to COM12

press Ctrl-C to enter the REPL
```

freemansoft, 2023-02-09:
Looks like we were troubleshooting at the same time.

BrianPugh, 2023-02-09:
I might have to search for 🐍 in the input stream; finally using an emoji in real logic!

freemansoft, 2023-02-09:
It look like "soft reboot" vanished

BrianPugh, 2023-02-09:
go ahead and give #95 a try. I'll merge it in later today. I'll be making a release (with a bunch of new package manager related features that have been merged into main) probably tomorrow.

freemansoft, 2023-02-09:
I'm on the circuitpython-8 branch in my local repo.  How do I pip install the branch version?

BrianPugh, 2023-02-09:
If you have it cloned locally and checked out, you should be able to run `pip install -e .` while in the root belay folder. The `-e` flag means to install it in "editable" mode, meaning that python just directly looks at that directory instead of installing a copy elsewhere in your system.

freemansoft, 2023-02-10:
No love but it might be my program.   I have a single one file that I converted to a two file host-and-mcu to run with belay and I get weird errors.  I'll try again another time

```
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
```

BrianPugh, 2023-02-10:
hmmm... I think the original issue is resolved (I both personally checked on real hardware with circuitpython 8.0.0, and I added circuitpython 8.0.0 to the integration tests and it passes). But i'll keep this issue open for now until you have some success.

freemansoft, 2023-02-10:
You can close it.  I'll probably wait until the next release to retest.

closed 2023-02-10 by freemansoft