BrianPugh/belay #94: CircuitPython remount in boot.py makes drive readonly to PC.

freemansoft, 2023-02-09:
I have this boot.py per the instructions and now the device is mounted read only to my pc.  I think it changes the drive from writable via circuitpython and read only to the computer.  You may wish to add notes on how to remove that `boot.py` from the repl so that people can code with files on the drive again.

- https://docs.circuitpython.org/en/latest/shared-bindings/storage/index.html
- https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage
- 

```
import storage

storage.remount("/")
```

BrianPugh, 2023-02-09:
I'll add documentation to the circuitpython section with the commands you provided, as well as instructions how to delete the ``boot.py``

freemansoft, 2023-02-10:
```
import os
os.listdir()
os.remove("boot.py")
```

BrianPugh, 2023-02-10:
I added more documentation in #96. Closing this for now, if you have feedback please provide it in the PR. I'll be merging it after I update some other documentation sections on new features.

closed 2023-02-10 by BrianPugh