BrianPugh/belay #100: global scope vars can't be accessed if reassigned in task

freemansoft, 2023-02-10:
I can print a global scope var in a function.  But I can't print it or access it if the variable is reassigned inside the function.  It looks like it is changing from the global scoped variable to a local scoped one.

The only variable that is important for this issue is `current_tics`. Including everything else for completiness
```
    @Device.setup(
        autoinit=True
    )  # ``autoinit=True`` means method will be called during object creation.
    def setup():
        import board
        import touchio
        import usb_hid
        from adafruit_hid.keyboard import Keyboard
        from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
        from adafruit_hid.keycode import Keycode
        import neopixel

        keyboard = Keyboard(usb_hid.devices)
        keyboard_layout = KeyboardLayoutUS(keyboard)

        touch1 = touchio.TouchIn(board.TOUCH1)
        touch2 = touchio.TouchIn(board.TOUCH2)

        pixels = neopixel.NeoPixel(board.NEOPIXEL, 4)

        default_color = (1, 1, 1)  # color when button not pressed
        current_color = default_color  # button color or default_color
        cycle_length_tics = 600  # loop cycle time --> blink cycle length
        cycle_blank_length_tics = cycle_length_tics // 3  # blanking time
        current_tics = 0
```

`print(current_tics)` fails here
```
    @Device.task
    def broken_vars():
        print("====try_touch()============")
        print_vars()
        print(keyboard)
        print(current_tics)
        current_tics = 0
```

`print(current_tics)` succeeds here
```
    @Device.task
    def broken_vars():
        print("====try_touch()============")
        print_vars()
        print(keyboard)
        print(current_tics)
```

failure message is 
> NameError: local variable referenced before assignment

It looks like the global scoped variable is replaced with a local one if I reassign it.  This means I can't have any loop counters or state between tasks.

freemansoft, 2023-02-10:
Hmmm.  This may be expected behavior?   If it is then I'm guessing everything needs to be passed as parameters and returned across each call.

BrianPugh, 2023-02-10:
This is just a generic python quirk:

```
MicroPython v1.19.1-628-gecd4d5439 on 2022-11-08; darwin [GCC 4.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> current_tics = 0
>>>
>>> def increment_tics():
...     print(current_tics)
...     current_tics += 1
...
...
...
>>> increment_tics()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 2, in increment_tics
NameError: local variable referenced before assignment
```

What you want is:
```
>>> current_tics = 0
>>> def increment_tics():
...     global current_tics
...     print(current_tics)
...     current_tics += 1
...
...
...
>>> increment_tics()
0
>>> increment_tics()
1
>>> increment_tics()
2
```

Basically, in python, the following are immutable datatypes:
* Int, Float
* Tuple
* String, Bytes

By default, Inside the scope of a function, global objects references are read-only. You can mutate an object (like a generic object, or a dict, or a list), but you cannot overwrite it. That is, however, unless you declare which `global` variables you'll be using inside the function.