BrianPugh/belay #97: ImportError no module named analogio

freemansoft, 2023-02-10:
i'm cool if belay is really targeted at the Pico RP 2040. But just in case

Firmware: Circuitpython 7.3
Board: Trinkey Neo 
Error: no module named 'analogio'

This module does not exist for this distribution of circuitpython on this board.

Code 
```
import argparse
import time

import belay

parser = argparse.ArgumentParser()
parser.add_argument("--port", "-p", default="/dev/ttyUSB0")
args = parser.parse_args()

# Setup the connection with the micropython board.
# This also executes a few common imports on-device.
device = belay.Device(args.port)
```

Outout
```
Traceback (most recent call last):
  File "C:\Users\joe\Documents\GitHub\Adafruit-Trinkey-CircuitPython\Neo\capacitive-touch-belay\standalone.py", line 12, in <module>
    device = belay.Device(args.port)
  File "C:\Users\joe\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\belay\device.py", line 275, in __init__
    self._exec_snippet("convenience_imports_circuitpython")
  File "C:\Users\joe\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\belay\device.py", line 348, in _exec_snippet
    return self("\n".join(snippets))
  File "C:\Users\joe\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\belay\device.py", line 410, in __call__
    self._board.exec(cmd, data_consumer=data_consumer)
  File "C:\Users\joe\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\belay\pyboard.py", line 524, in exec
    raise PyboardException(ret_err.decode())
belay.pyboard.PyboardException:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'analogio'
```

I think it is because because `convenience_imports_circuitpython.py` contains modules that are not on all boards.
```
import os, time, analogio, board, digitalio
from time import sleep
from busio import I2C, SPI
```

I feel like a bad penny

BrianPugh, 2023-02-10:
The goal for Belay is to be as compatible with as many boards/ports as practically possible.

I didn't realize that `analogio` isn't available on all boards; certainly makes sense for microcontrollers that don't have a builtin ADC. I'll have a bug fix in shortly, I'm going to assemble a list of modules that might not be available in every port and only import them if they're available. 

Keep opening these issues! I really appreciate them; a library only grows strong and robust as people use it :D .

BrianPugh, 2023-02-10:
Should be resolved in #98, will merge soon; expect a release tomorrow.

closed 2023-02-10 by BrianPugh

BrianPugh, 2023-02-10:
I honestly do too. I agree they can be confusing, but they also make a lot of basic use-cases and demos much less kludgy (especially before the class-interface was developed).

For purity, you can disable convenience imports by providing an empty string `startup` argument when instantiating a device:

```
device = Device("COM4", startup="")  # Performs no convenience imports
```