BrianPugh/belay #113: Bug: `setup` with arguments fails

roaldarbol, 2023-03-13:
On `belay` 0.15, I could do:
```python
from belay import Device, list_devices

class Pico(Device):
    @Device.setup
    def setup1(argument=False):
        from machine import Pin
        led = Pin(25, Pin.OUT)

    @Device.task
    def led_toggle():
        led.toggle()

if __name__ == "__main__":
    port = list_devices()[-1]
    with Pico(port) as pico:
        pico.setup1(argument=True)
        pico.led_toggle()
```
However, I just updated to 0.19, and it no longer works - for good reason of course, as I can't provide an argument to stuff that `autoinit`s. However, even adding `autoinit=False` gives the error:
```python
from belay import Device, list_devices

class Pico(Device):
    @Device.setup(
        autoinit=False
    )
    def setup1(argument=False):
        from machine import Pin
        led = Pin(25, Pin.OUT)

    @Device.task
    def led_toggle():
        led.toggle()

if __name__ == "__main__":
    port = list_devices()[-1]
    with Pico(port) as pico:
        pico.setup1(argument=True)
        pico.led_toggle()

# ValueError: Method <function Pico.setup1 at 0x10699a320> decorated with "@Device.setup(autoinit=True)" must have no arguments.
```

BrianPugh, 2023-03-13:
I am able to reproduce, sorry about that! will have a fix out soon.

roaldarbol, 2023-03-13:
As always, no worries at all! I hadn't noticed that I had specified `^0.15` in my `pyproject.toml`, so actually hadn't received any of the new sauce yet. :-P

BrianPugh, 2023-03-13:
Will merge and tag a release once CI is done doing it's thing.

BrianPugh, 2023-03-13:
should be fixed in the newly released `v0.19.1` (not to be confused with micropython version 🙈)

closed 2023-03-13 by BrianPugh