BrianPugh/belay #148: Can one return bytes or bytearray from task decorated function?

pdietl, 2023-09-12:
Just a question here. Can I return a bytearray or bytes from a @device.task decorated function? I tried but I'm getting some errors:

```
bytearray(b'Hello, world!')
<class 'bytearray'>
Traceback (most recent call last):
  File "/home/pdietl/unorg/mpy-belay/./demo.py", line 93, in <module>
    chip_read(0, len(msg))
  File "/home/pdietl/unorg/mpy-belay/.venv/lib/python3.10/site-packages/belay/executers.py", line 141, in func_executer
    return self._belay_device._traceback_execute(src_file, src_lineno, name, cmd, record=record)
  File "/home/pdietl/unorg/mpy-belay/.venv/lib/python3.10/site-packages/belay/device.py", line 790, in _traceback_execute
    res = self(cmd, record=record)
  File "/home/pdietl/unorg/mpy-belay/.venv/lib/python3.10/site-packages/belay/device.py", line 286, in __call__
    self._board.exec(cmd, data_consumer=data_consumer)
  File "/home/pdietl/unorg/mpy-belay/.venv/lib/python3.10/site-packages/belay/pyboard.py", line 592, in exec
    ret, ret_err = self.exec_raw(command, data_consumer=data_consumer)
  File "/home/pdietl/unorg/mpy-belay/.venv/lib/python3.10/site-packages/belay/pyboard.py", line 584, in exec_raw
    return self.follow(timeout, data_consumer)
  File "/home/pdietl/unorg/mpy-belay/.venv/lib/python3.10/site-packages/belay/pyboard.py", line 506, in follow
    data = self.read_until(b"\x04", timeout=timeout, data_consumer=data_consumer)
  File "/home/pdietl/unorg/mpy-belay/.venv/lib/python3.10/site-packages/belay/pyboard.py", line 440, in read_until
    data_consumer(data_for_consumer)
  File "/home/pdietl/unorg/mpy-belay/.venv/lib/python3.10/site-packages/belay/device.py", line 280, in data_consumer
    out = parse_belay_response(line)
  File "/home/pdietl/unorg/mpy-belay/.venv/lib/python3.10/site-packages/belay/device.py", line 61, in parse_belay_response
    return ast.literal_eval(line)
  File "/usr/lib/python3.10/ast.py", line 110, in literal_eval
    return _convert(node_or_string)
  File "/usr/lib/python3.10/ast.py", line 109, in _convert
    return _convert_signed_num(node)
  File "/usr/lib/python3.10/ast.py", line 83, in _convert_signed_num
    return _convert_num(node)
  File "/usr/lib/python3.10/ast.py", line 74, in _convert_num
    _raise_malformed_node(node)
  File "/usr/lib/python3.10/ast.py", line 71, in _raise_malformed_node
    raise ValueError(msg + f': {node!r}')
ValueError: malformed node or string on line 1: <ast.Call object at 0x7ffb4422ca60>
```

The first print is a `print` of the bytearray I want to return.

BrianPugh, 2023-09-12:
Currently we don't support `bytearray`s (we do support `bytes`, however). Belay works on the symmetry of `repr` serialization and `ast.literal_eval` deserialization to work. However, I just noticed that `ast.literal_eval('b"\x00"')` doesn't work (`literal_eval` doesn't like null bytes), so maybe we should swap it out with a more powerful parser.

BrianPugh, 2023-09-12:
If we swap out `ast.literal_eval` with the unsafe `eval`, we can support `bytearray`s (as well as any other class who's `repr` perfectly matches it's constructor). However, this then allows arbitrary execution from a remote device, which doesn't sound like a great idea.

Maybe we can set an "unsafe parsing" boolean that then sets whether or not to use `ast.literal_eval` vs `eval`. Thoughts?

pdietl, 2023-09-12:
I like the second approach.

BrianPugh, 2023-09-13:
this should be pretty easy; i'll really try to get to this and the pydantic stuff soon.

BrianPugh, 2023-09-25:
closing this issue, further discussion can be done in the PR. I'll be merging and releasing once CI is sorted out (it's like 50% sorted out as of right now).

closed 2023-09-25 by BrianPugh