BrianPugh/belay #150: How to control memory allocation

pdietl, 2023-09-13:
So I am calling a function that writes its argument of bytes to a SPI FRAM chip. I call it with a bytes argument of say, 10k bytes. When the function call is over, even when I call `gc.collect()`, the memory in microPython keeps going down. Any suggestions?

My function looks like this:
```
@device.task
def chip_write_remote(bytes_to_write):
    print(gc.mem_free())
    gc.collect()
    # Write enable
    cs(0)
    spi.write(b'\x06')
    cs(1)
    # Perform the write
    cs(0)
    spi.write(bytes_to_write)
    cs(1)
    del bytes_to_write
    print(gc.mem_free())
    gc.collect()
    print(gc.mem_free())
```

BrianPugh, 2023-09-13:
In a nutshell, what happens is the following:

1. Belay sends over the source code of the `chip_write_remote` function over to micropython.
2. Belay asks micropython to evaluate the expression `f"print('_BELAYR' + repr({cmd}))"`, where cmd is `chip_write_remote(b'ABC')`. This quickly gets confusing, since we have python-code, python-code-as-string, and python-code-subsitution-in-string. Simplifying, it asks micropython to evaluate `chip_write_remote(b'ABC')`, get the string representation of the response, and then print it to stdout (where host-belay then parses the string representation back into a python object).

All of this is to say, that despite deleting the reference to your variable `bytes_to_write` inside your function, it won't get garbage collected until all references are deleted. In this case, the calling context (the `print` expression) still has a reference to `bytes_to_write`. That quickly goes out of scope once the results are received by host-belay. I think if you called `gc.collect()` in a different belay task, you would see the memory usage go down.

Are you sure you need to manually garbage collect? Micropython should just handle it for you automatically.

pdietl, 2023-09-13:
Hmm. Thank you for the explanation. Based on that description, I would expect all references to `bytes_to_write` to be gone right after calling `chip_write_remote` at the end of the loop, right before I call `device(...)`. At that point the empty response has been printed and parsed on my Linux machine, right? Unless I'm missing something.

BrianPugh, 2023-09-15:
I think your expectations are correct, I can experiment around with this this weekend.

BrianPugh, 2023-09-16:
this all works fine for me; complete script below. Note; your `write_buffer.extend(data_bytes)` should be `write_buffer.extend(bites)`

```
from belay import Device
from itertools import islice

device = Device()

@device.setup
def imports():
    import gc

@device.task
def chip_write_remote(bytes_to_write):
    print(gc.mem_free())
    gc.collect()
    del bytes_to_write
    print(gc.mem_free())
    gc.collect()
    print(gc.mem_free())

def batched(iterable, n):
    "Batch data into tuples of length n. The last batch may be shorter."
    # batched('ABCDEFG', 3) --> ABC DEF G
    if n < 1:
        raise ValueError('n must be at least one')
    it = iter(iterable)
    while batch := bytes(islice(it, n)):
        yield batch

def chip_write(address, data_bytes):
    assert address >> 18 == 0, "Address must be 18 bits or less!"

    for bites in batched(data_bytes, 50*1024):
        write_buffer = bytearray()
        write_buffer.append(0x02)
        write_buffer.extend(address.to_bytes(3, byteorder='big'))
        write_buffer.extend(bites)
        chip_write_remote(write_buffer)
        device("import gc; gc.collect()")

def main():
    imports()
    data = bytes(200 * 1024)
    chip_write(0x10, data)

if __name__=="__main__":
    main()
```

output:
```
python scratch.py
117136
119360
119360
119168
119344
119344
119168
119344
119344
119168
119344
119344
```

pdietl, 2023-09-20:
@BrianPugh 

1. Thank you so much for helping me out! I really appreciate it.

2. That mistake that you found was the entire problem. I was trying to _not_ send all of the bytes in  `data_bytes` because I knew it wouldn't fit in memory. So I completely shot myself in the foot 🫠.

3. Do you think an example like this would be interesting to add to the documentation somewhere?

closed 2023-09-20 by pdietl