BrianPugh/belay #155: LED Example  does not run

W-Michael, 2023-10-31:
I am using desktop with an AMD Athlon(tm) II X4 635 Processor the OS is Linux Mint 20.2.
Thonny is installed and my Raspberry PI Pico (with MicroPytho) is connected. In Thonny I can turn the onboard LED on and off, and can get readings from the onboard temperature sensor but cannot get the belay example to run.
The following is my version for the LED example (pico_led.py) annotated to show my configuration and the output when it is run from the terminal:
import belay

# list_ports_linux.py
# /dev/ttyACM0 - Board in FS mode - Board CDC: usb

# udevadm info -q all -a -n /dev/ttyACM0
#  looking at parent device '/devices/pci0000:00/0000:00:13.1/usb6/6-2/6-2:1.0'
#  ATTRS{interface}=="Board CDC"
#  looking at parent device '/devices/pci0000:00/0000:00:13.1/usb6/6-2':
#  ATTRS{manufacturer}=="MicroPython"
#  ATTRS{product}=="Board in FS mode"

# lsusb
# Bus 006 Device 002: ID 2e8a:0005 MicroPython Board in FS mode
device = belay.Device("/dev/ttyACM0")
from machine import Pin

@device.task
def set_led(state):
    print(f"Printing from device; turning LED to {state}.")
    Pin(25, Pin.OUT).value(state)

set_led(True)
''' The following is copied from the terminal
python3 pico_led.py
Traceback (most recent call last):
  File "pico_led.py", line 14, in <module>
    device = belay.Device("/dev/ttyACM0")
  File "/home/user1/.local/lib/python3.8/site-packages/belay/device.py", line 120, in __init__
    self._exec_snippet("startup")
  File "/home/user1/.local/lib/python3.8/site-packages/belay/device.py", line 239, in _exec_snippet
    snippets = [read_snippet(name) for name in names]
  File "/home/user1/.local/lib/python3.8/site-packages/belay/device.py", line 239, in <listcomp>
    snippets = [read_snippet(name) for name in names]
  File "/home/user1/.local/lib/python3.8/site-packages/belay/helpers.py", line 23, in read_snippet
    return importlib.resources.files(snippets).joinpath(resource).read_text()
AttributeError: module 'importlib.resources' has no attribute 'files'
'''
Are there fixes for the scripts in belay
Thanks
Michael

BrianPugh, 2023-10-31:
what version of python are you running?

W-Michael, 2023-11-01:
Hi Brian
I am running Python 3.8.10

Michael

closed 2023-11-26 by BrianPugh