BrianPugh/belay #167: Strings in lists defined in a @Device.setup function can be replaced with the number 0.

adeuring, 2024-08-17:
The title may sound odd but the following script gives really weird output:

```
from belay import Device

device = Device('/dev/ttyUSB5')

@device.setup
def setup():
    l1 = [
        'foo', 'bar']
    l2 = ['foo', 'bar']
    l3 = ['foo',
          'bar']
    l4 = ['foo',
          'bar',
          'baz']
    l5 = [
        'foo', 'foo',
        'bar', 'bar',
        'baz', 'baz']
    print('l1', l1)
    print('l2', l2)
    print('l3', l3)
    print('l4', l4)
    print('l5', l5)

setup()
```

Result:

```
l1 [0, 'bar']
l2 ['foo', 'bar']
l3 ['foo', 0]
l4 ['foo', 0, 0]
l5 [0, 'foo', 0, 'bar', 0, 'baz']
```

So, the first string in a line is replaced with the integer 0.
Seen with Belay 0.22.3 and Micropython v1.20.0 on an ESP32.

The error does not occur in a task.

adeuring, 2024-08-17:
Oh, similar things happen for dicitonaries and sets defined in a setup function:

```
    d = {
        'a': 42,
        'b': 43,
        1: 44}
    print(d)

    s = {
        'a',
        'b',
        'c',
        }
    print(s)
```

gives this output:

```
{0: 43, 1: 44}
{0}
```

adeuring, 2024-08-17:
Found a workaround: Define the list in a function nested inside the setup function:

```
@device.setup
def setup():

    def make_list():
        return [
            'a1', 'a2',
            'b1', 'b2',
            ]

    l = make_list()
    print(l)
```

Result:

```
['a1', 'a2', 'b1', 'b2']
```

BrianPugh, 2024-08-18:
Hi @adeuring! Thank you for opening this issue; I am able to reproduce in my setup. There is a bug in my code-minifying function. Investigating now!

BrianPugh, 2024-08-18:
Fixed in v0.22.4

closed 2024-08-18 by BrianPugh

adeuring, 2024-08-18:
Wow, thanks, that was really fast!