BrianPugh/belay #176: Citation

roaldarbol, 2025-02-23:
Hi @BrianPugh! 
It's finally nearing my PhD submission!!! I would love to give a proper citation to Belay as it was foundational to much of my data collection. Could you consider adding a `CITATION.cff` file to the project? You can see more about it (it's not complicated at all) and even create one on https://citation-file-format.github.io/. 

I'm submitting on Friday, so if it's possible to have it before then it would be amazing, but otherwise I'll make a custom citation to the project myself. :-) 

(You might also find your name in my acknowledgements - thanks for so many good discussions - it's always been a pleasure to work with you!)
Cheers, Mikkel

BrianPugh, 2025-02-23:
Congratulations on wrapping up your PhD! And thanks for mentioning me/Belay in your paper, its totally unnecessary, but very much welcome 😄 .

Please post a link here if/when your work becomes publicly available!

roaldarbol, 2025-02-23:
Thanks! Will do! 😄

closed 2025-02-23 by BrianPugh