BrianPugh/belay #190: support disabling nativemodule

jsiverskog, 2025-08-19:
we have a number of boards on the field running `3.4.0; MicroPython v1.23.0-preview.168.g578365e7b on 2024-02-28` which with belay 0.25.0 emits this error:

```
belay.pyboard.PyboardException: 

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: incompatible .mpy file
```

sure it's an (old) preview version, which probaby causes this, and the correct long-term solution is to update those boards to something newer (micropython 1.25.0 works), but it would be neat if there would be a way to disable the native module somehow.

jsiverskog, 2025-08-19:
or is it possible to detect the error above and fallback?

BrianPugh, 2025-08-19:
I'm thinking we should detect the error and fallback; what's the full stack trace?

jsiverskog, 2025-08-19:
something like this (the full stacktrace is several pages long but contains mostly unrelated stuff):
```
  File "XXX.py", line 50, in func
    self._device.sync("ZZZ")
  File "venv/lib/python3.11/site-packages/belay/device.py", line 466, in sync
    self._exec_snippet(*snippets_to_execute)
  File "venv/lib/python3.11/site-packages/belay/device.py", line 271, in _exec_snippet
    return self("\n".join(snippets))
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "venv/lib/python3.11/site-packages/belay/device.py", line 343, in __call__
    self._board.exec(cmd, data_consumer=data_consumer)
  File "venv/lib/python3.11/site-packages/belay/pyboard.py", line 605, in exec
    raise PyboardException(ret_err.decode())
belay.pyboard.PyboardException: 

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: incompatible .mpy file
```