BrianPugh/belay #196: Timeout / clogging?

11mbs, 2025-08-28:
Hi, thanks for the awesome code!

I've noticed an issue where after a certain number of calls to the device, it seems to seize up and time out.
I have found and modified the timeout value but the outcome is the same (if no timeout, then no progress).
The result does seem to be repeatable and consistent, however - it always stops at the same point (after 5 calls post-setup).  
Any ideas on what might be causing this?
I'm certain I'm not creating new objects each call.

11mbs, 2025-08-28:
I'm using an esp32 to control the motors + endstops on a moving microscope stage. 
I'm using belay to automate sending the movement commands in a script that I use to take photos with a usb microscope.  
Basically a for loop of move -> image -> repeat.
What I'm seeing on the terminal is this:

`---------------------------------------------------------------------------
PyboardError                              Traceback (most recent call last)
File ~/Documents/GitHub/OF-larger-delta-stage/code/data_collection/Movement Reliability Data Collection.py:74, in <module>
     72 stage = usb.find_usb_device(False)
     73 stage("stage.zero(True)")
---> 74 move_collect(positions,folder_path,backlash=50)
     75 stage.close()
     76 end_time = int(time.time())

File ~/Documents/GitHub/OF-larger-delta-stage/code/data_collection/Movement Reliability Data Collection.py:35, in move_collect(positions, folder_path, backlash, i, rest)
     31 for position in positions:
     32     #reset_usb_device.reset_usb_device()
     33     # initial image
     34     pos = [0,0,0]
---> 35     stage(f"stage.move_to({pos})")
     36     filepath = save_path + str(i).zfill(3) + f'_x{pos[0]}_y{pos[1]}_z{pos[2]}.raw'
     37     time.sleep(1)

File ~/imagecapture/lib/python3.12/site-packages/belay/device.py:314, in Device.__call__(self, cmd, minify, stream_out, record, trusted)
    311                 stream_out.write(line)
    313 try:
--> 314     self._board.exec(cmd, data_consumer=data_consumer)
    315 except (SerialException, ConnectionResetError) as e:
    316     # Board probably disconnected.
    317     if self.attempts:

File ~/imagecapture/lib/python3.12/site-packages/belay/pyboard.py:589, in Pyboard.exec(self, command, data_consumer)
    588 def exec(self, command, data_consumer=None):
--> 589     ret, ret_err = self.exec_raw(command, data_consumer=data_consumer)
    590     if ret_err:
    591         raise PyboardException(ret_err.decode())

File ~/imagecapture/lib/python3.12/site-packages/belay/pyboard.py:580, in Pyboard.exec_raw(self, command, timeout, data_consumer)
    579 def exec_raw(self, command, timeout=None, data_consumer=None):
--> 580     self.exec_raw_no_follow(command)
    581     return self.follow(timeout, data_consumer)

File ~/imagecapture/lib/python3.12/site-packages/belay/pyboard.py:564, in Pyboard.exec_raw_no_follow(self, command)
    561     return self.raw_paste_write(command_bytes)
    562 else:
    563     # Device doesn't support raw-paste, fall back to normal raw REPL.
--> 564     self.read_until(b"w REPL; CTRL-B to exit\r\n>")
    565 # Don't try to use raw-paste mode again for this connection.
    566 self.use_raw_paste = False

File ~/imagecapture/lib/python3.12/site-packages/belay/pyboard.py:461, in Pyboard.read_until(self, ending, timeout, data_consumer)
    458 while not self._unconsumed_buf:
    459     # loop until new data has arrived.
    460     if time.time() > deadline:
--> 461         raise PyboardError(
    462             f"Timed out reading until {repr(ending)}\n    Received: {repr(self._consumed_buf)}"
    463         )
    465     if not self.serial.in_waiting:
    466         time.sleep(0.001)

PyboardError: Timed out reading until b'w REPL; CTRL-B to exit\r\n>'
    Received: bytearray(b'0\r\nB:1\r\nC:1\r\nR\x01\x80\x00\x01')`

I'll work on making a version of the esp32 code that doesn't need to be connected to anything physically.

BrianPugh, 2025-08-28:
it could be possible that something we're doing doesn't properly respect the [flow-control with raw-mode](https://docs.micropython.org/en/latest/reference/repl.html#raw-mode-and-raw-paste-mode), but I would be surprised in this case.

Is it possible that turning on/off the motor is introducing noise/power-sag in your power supply, causing funky behavior from the esp32? Maybe try disconnecting it or replacing it with an LED or something and see if this problem persists.